xreef/ABB_Aurora_Solar_Inverter_Library #4: no result from readDSP

fencer, 2021-08-22:
I am implementing your library within my ESPHome application to connect to my ABB PVI-3.0-TL-OUTD-S. The readCumulatedEnergy seems to work (although not use the values are correct.) however the readDSP seems not to to work and state.readState is always 0. Any pointers on what should I check to get it working?

xreef, 2021-08-22:
Hi,
you can find  information on that method here
https://www.mischianti.org/2020/08/20/abb-aurora-pv-inverter-library-for-arduino-esp8266-and-esp32/#Measure_request_to_the_DSP
Bye Renzo

fencer, 2021-08-22:
Yes that is the article I am using as a reference however the data seems not to be returned
[image]

xreef, 2021-08-23:
Hi,
add your code to this issue and we are going to check.
Bye Renzo

fencer, 2021-08-23:
Code below. There is some code which is specific to ESPHome and HomeAssistant.

`#include <Aurora.h>
#include "esphome.h"
#define RED_LED_PIN 13
#define LED_PIN 12

Aurora inverter = Aurora(2, 3, 1, 5);

class AuroraSensor : public PollingComponent {
public:
TextSensor *DcDcChannel1State_sensor = new TextSensor();
TextSensor *DateTime_sensor = new TextSensor();
TextSensor *SerialNumber_sensor = new TextSensor();
TextSensor *InverterName_sensor = new TextSensor();

Sensor *PowerIn1_sensor = new Sensor();
Sensor *PowerIn2_sensor = new Sensor();
Sensor *GridPower_sensor = new Sensor();
Sensor *TemperatureInverter_sensor = new Sensor();
Sensor *TemperatureBooster_sensor = new Sensor();
Sensor *DayCumulatedEnergy_sensor = new Sensor();
Sensor *VIn1_sensor = new Sensor();
Sensor *IIn1_sensor = new Sensor();
Sensor *VIn2_sensor = new Sensor();
Sensor *IIn2_sensor = new Sensor();
Sensor *IsolationResistance_sensor = new Sensor();
Sensor *PowerPeak_sensor = new Sensor();
Sensor *PowerPeakToday_sensor = new Sensor();

AuroraSensor() : PollingComponent(60000) { }

void setup() override
{
    pinMode(RED_LED_PIN, OUTPUT);
    pinMode(LED_PIN, OUTPUT);
	inverter.begin();
	ESP_LOGD("Aurora", "Setup ok");
	digitalWrite(LED_PIN, HIGH);
}

void update() override
{
    digitalWrite(RED_LED_PIN, HIGH);
    
    ESP_LOGD("Aurora", "-- Init update data --");
 

    Aurora::DataSystemSerialNumber dataSystemSerialNumber = inverter.readSystemSerialNumber();
    SerialNumber_sensor->publish_state(dataSystemSerialNumber.SerialNumber.c_str());

    Aurora::DataTimeDate dataTimeDate = inverter.readTimeDate();
    DateTime_sensor->publish_state(dataTimeDate.getDateTimeString().c_str());

    Aurora::DataVersion dataVersion = inverter.readVersion();
    InverterName_sensor->publish_state(dataVersion.getModelName().name.c_str());

    Aurora::DataCumulatedEnergy cumulatedEnergy = inverter.readCumulatedEnergy(CUMULATED_DAILY_ENERGY);
    DayCumulatedEnergy_sensor->publish_state(cumulatedEnergy.energy);

    Aurora::DataDSP VIn1dataDSP = inverter.readDSP(DSP_VOLTAGE_ALL);
    ESP_LOGD("Aurora", "Read state -->  %d",VIn1dataDSP.state.readState);
    if (VIn1dataDSP.state.readState == 1) {
        ESP_LOGD("Aurora", "VIn1dataDSP: %s",VIn1dataDSP.value);
        VIn1_sensor->publish_state(VIn1dataDSP.value);
    }

    Aurora::DataDSP VIn2dataDSP = inverter.readDSP(DSP_VOLTAGE_GT);
    if (VIn2dataDSP.state.readState == 1) {
        VIn2_sensor->publish_state(VIn2dataDSP.value);
    }

    Aurora::DataDSP IIn1dataDSP = inverter.readDSP(DSP_CURRENT_ALL);
    if (IIn1dataDSP.state.readState == 1) {
        IIn1_sensor->publish_state(IIn1dataDSP.value);
    }

    Aurora::DataDSP IIn2dataDSP = inverter.readDSP(DSP_CURRENT_GT);
    if (IIn2dataDSP.state.readState == 1) {
        IIn2_sensor->publish_state(IIn2dataDSP.value);
    }
    
    Aurora::DataDSP TemperatureInverterdataDSP = inverter.readDSP(DSP_INVERTER_TEMPERATURE_GT);
    if (TemperatureInverterdataDSP.state.readState == 1) {
        TemperatureInverter_sensor->publish_state(TemperatureInverterdataDSP.value);
    }

    Aurora::DataDSP TemperatureBoosterdataDSP = inverter.readDSP(DSP_BOOSTER_TEMPERATURE_GT);
    if (TemperatureBoosterdataDSP.state.readState == 1) {
        TemperatureBooster_sensor->publish_state(TemperatureBoosterdataDSP.value);
    }

    Aurora::DataDSP PowerIn1dataDSP = inverter.readDSP(DSP_PIN1_ALL);
    if (PowerIn1dataDSP.state.readState == 1) {
        PowerIn1_sensor->publish_state(PowerIn1dataDSP.value);
    }

    Aurora::DataDSP PowerIn2dataDSP = inverter.readDSP(DSP_PIN2);
    if (PowerIn2dataDSP.state.readState == 1) {
        PowerIn2_sensor->publish_state(PowerIn2dataDSP.value);
    }

    Aurora::DataDSP GridPowerdataDSP = inverter.readDSP(DSP_GRID_POWER_ALL);
    if (GridPowerdataDSP.state.readState == 1) {
        GridPower_sensor->publish_state(GridPowerdataDSP.value);
    }

    Aurora::DataDSP IsolationResistancedataDSP = inverter.readDSP(DSP_ISOLATION_RESISTANCE_RISO_ALL);
    if (IsolationResistancedataDSP.state.readState == 1) {
        IsolationResistance_sensor->publish_state(IsolationResistancedataDSP.value);
    }

    Aurora::DataDSP PowerPeakdataDSP = inverter.readDSP(DSP_POWER_PEAK_ALL);
    if (PowerPeakdataDSP.state.readState == 1) {
        PowerPeak_sensor->publish_state(PowerPeakdataDSP.value);
    }

    Aurora::DataDSP PowerPeakTodaydataDSP = inverter.readDSP(DSP_POWER_PEAK_TODAY_ALL);
    if (PowerPeakTodaydataDSP.state.readState == 1) {
        PowerPeakToday_sensor->publish_state(PowerPeakTodaydataDSP.value);
    }

    ESP_LOGD("Aurora", "-- Done update data --");

    digitalWrite(RED_LED_PIN, LOW);
}
};`

fencer, 2021-08-23:
ok so I updated the code and this is the output!!

```
#include <Aurora.h>
#include "esphome.h"
#define RED_LED_PIN 13
#define LED_PIN 12

Aurora inverter = Aurora(2, 3, 1, 5);

class AuroraSensor : public PollingComponent {
public:
TextSensor *DcDcChannel1State_sensor = new TextSensor();
TextSensor *DateTime_sensor = new TextSensor();
TextSensor *SerialNumber_sensor = new TextSensor();
TextSensor *InverterName_sensor = new TextSensor();

Sensor *PowerIn1_sensor = new Sensor();
Sensor *PowerIn2_sensor = new Sensor();
Sensor *GridPower_sensor = new Sensor();
Sensor *TemperatureInverter_sensor = new Sensor();
Sensor *TemperatureBooster_sensor = new Sensor();
Sensor *DayCumulatedEnergy_sensor = new Sensor();
Sensor *WeekCumulatedEnergy_sensor = new Sensor();
Sensor *SevenDayCumulatedEnergy_sensor = new Sensor();
Sensor *MonthCumulatedEnergy_sensor = new Sensor();
Sensor *YearCumulatedEnergy_sensor = new Sensor();
Sensor *LifetimeCumulatedEnergy_sensor = new Sensor();
Sensor *SinceResetEnergy_sensor = new Sensor();
Sensor *VIn1_sensor = new Sensor();
Sensor *IIn1_sensor = new Sensor();
Sensor *VIn2_sensor = new Sensor();
Sensor *IIn2_sensor = new Sensor();
Sensor *IsolationResistance_sensor = new Sensor();
Sensor *PowerPeak_sensor = new Sensor();
Sensor *PowerPeakToday_sensor = new Sensor();

AuroraSensor() : PollingComponent(60000) { }

void setup() override
{
    pinMode(RED_LED_PIN, OUTPUT);
    pinMode(LED_PIN, OUTPUT);
	inverter.begin();
	ESP_LOGD("Aurora", "Setup ok");
	digitalWrite(LED_PIN, HIGH);
}

void update() override
{
    digitalWrite(RED_LED_PIN, HIGH);
    
    ESP_LOGD("Aurora", "-- Init update data --");
 

    Aurora::DataSystemSerialNumber dataSystemSerialNumber = inverter.readSystemSerialNumber();
    SerialNumber_sensor->publish_state(dataSystemSerialNumber.SerialNumber.c_str());

    Aurora::DataTimeDate dataTimeDate = inverter.readTimeDate();
    DateTime_sensor->publish_state(dataTimeDate.getDateTimeString().c_str());

    Aurora::DataVersion dataVersion = inverter.readVersion();
    InverterName_sensor->publish_state(dataVersion.getModelName().name.c_str());

    // ------------- Energy Sensors ------------------

    Aurora::DataCumulatedEnergy cumulatedDayEnergy = inverter.readCumulatedEnergy(CUMULATED_DAILY_ENERGY);
    DayCumulatedEnergy_sensor->publish_state(cumulatedDayEnergy.energy);

    Aurora::DataCumulatedEnergy cumulatedWeekEnergy = inverter.readCumulatedEnergy(CUMULATED_WEEKLY_ENERGY);
[... 66 lines cut ...]
    }

    Aurora::DataDSP IsolationResistancedataDSP = inverter.readDSP(DSP_ISOLATION_RESISTANCE_RISO_ALL);
    if (IsolationResistancedataDSP.state.readState == 1) {
        IsolationResistance_sensor->publish_state(IsolationResistancedataDSP.value);
    }

    Aurora::DataDSP PowerPeakdataDSP = inverter.readDSP(DSP_POWER_PEAK_ALL);
    if (PowerPeakdataDSP.state.readState == 1) {
        PowerPeak_sensor->publish_state(PowerPeakdataDSP.value);
    }

    Aurora::DataDSP PowerPeakTodaydataDSP = inverter.readDSP(DSP_POWER_PEAK_TODAY_ALL);
    if (PowerPeakTodaydataDSP.state.readState == 1) {
        PowerPeakToday_sensor->publish_state(PowerPeakTodaydataDSP.value);
    }

    ESP_LOGD("Aurora", "-- Done update data --");

    digitalWrite(RED_LED_PIN, LOW);
}
};
```

[image]

xreef, 2021-08-29:
Hi fencers,
the coffee seems good, check my code that work with my inverter.
https://github.com/xreef/Aurora_Web_Invert_Monitor/blob/056c5ecba246478ee3b7c47151d0cda3456b2dc0/Aurora-Web-Invert-Monitor.ino
Bye Renzo