xreef/ABB_Aurora_Solar_Inverter_Library #5: Communication problem

stefanopradella, 2022-04-06:
Hello, I am trying to read the power output of my inverter (ABB PVI-3.0-TL-OUTD) and I flashed your example code as it is,  changing the pin mapping.
The only change I made to the code is to replace the calls to freeMemory with ESP.getFreeHeap(), as for some reason the code is not compiling for me on the ESP if using the MemoryFree library.

I am using a Wemos D1 mini board, connected to an RS485 to TTL Arduino board (MAX485) through a level shifter.
Connections are made as following:
```
[Board to module]
rxPin (0) -> RO
txPin (5) -> DI
sCCPin (4) -> RE, DE

[Module to inverter]
A -> +T/R
B -> -T/R
RTN -> GND
LNK disconnected
```
Output I'm getting on the console:
```
15:54:43.429 -> INVERTER 2
15:54:43.478 ->           Data ROW = 255 255 0 0 0 0 0 0  
15:54:43.478 ->         Read State = 0
15:54:43.478 -> Transmission State = Unknown
15:54:43.512 ->       Global State = Unknown
15:54:43.512 ->            Version = Unknown Unknown Unknown Unknown
15:54:43.545 -> ------------------------------------------
15:54:47.578 -> ------------------------------------------
15:54:47.578 -> INVERTER 2
15:54:47.578 ->           Data ROW = 255 255 0 0 0 0 0 0  
15:54:47.611 ->         Read State = 0
15:54:47.611 -> Transmission State = Unknown
15:54:47.645 ->       Global State = Unknown
15:54:47.645 ->            Energia = 0 Wh
15:54:47.678 -> ------------------------------------------
15:54:47.678 -> freeMemory()=47080
15:54:47.711 -> freeMemory()=47080
15:54:47.711 -> INVERTER 2
15:54:47.711 ->           Data ROW = 255 255 255 255 255 255 0 0  
15:54:47.745 ->         Read State = 0
15:54:47.745 -> Transmission State = Unknown
15:54:47.778 ->       Global State = Unknown
15:54:47.778 ->           Alarms 1 = Unknown
15:54:47.811 ->           Alarms 2 = Unknown
15:54:47.811 ->           Alarms 3 = Unknown
15:54:47.845 ->           Alarms 4 = Unknown
15:54:47.845 -> ------------------------------------------
```
The inverter address is correctly set and the terminator resistance is set to ON. 

I must be missing something, thank you for your help.

xreef, 2022-04-06:
Hi Stefano,
I don't know if the level shifter works correctly with the IC.
[I try with max3485 and work correctly with WeMos,](https://www.mischianti.org/2020/05/11/interface-arduino-esp8266-esp32-rs-485/) I don't know if that configuration can generate problems.
Bye Renzo

xreef, 2022-04-11:
no activiti

closed 2022-04-11 by xreef

gfucka, 2023-03-05:
@stefanopradella do you find a solution? I have same problem. Thanks

stefanopradella, 2023-03-05:
> @stefanopradella do you find a solution? I have same problem. Thanks

Unfortunately no, I gave up