xreef/ABB_Aurora_Solar_Inverter_Library #6: Meaning of the multipler in the inverter struct

epol, 2022-07-02:
Hi,

I see that in the InverterModel struct there is [a variable called `multipler`](https://github.com/xreef/ABB_Aurora_Solar_Inverter_Library/blob/55559b139a291a953a582200deb7f6de9ded7af7/include/statesNaming.h#L682). The values of this variables are constants defined in the following conversion function that takes the model ID and coverts it to the model name.

I'm not finding any reference to that values in the documentation, where does that numbers come from and what is their meaning?

Thank you and congrats for the great code!

xreef, 2022-07-04:
Hi @epol,
sorry epol I don't remember where I get those values, for the meaning I think It's well explained [here](https://manpages.ubuntu.com/manpages/bionic/man1/aurora.1.html).
Bye Renzo

epol, 2022-07-09:
Hi Renzo,

thank you for the reference, I've found the same variable and values on the software you mentioned (in the source code found on the [home page](http://www.curtronics.com/Solar/AuroraData.html).

Apparently Aurora reports some power values wrong, so that value is used to get the real value.

Also from the [stats published by the author](http://www.curtronics.com/Solar/CurStats.html) (see the notes) the power reported on the DSP variable *grid power* is different from the the power computed as the product of the DSP values *grid voltage* and *grid current*. But I haven't investigate this further yet.

Thank you again for the pointers to the original code and, of course, for your great Arduino code!

Cheers,
Enrico

xreef, 2022-07-11:
Oh yes, I remember something similar.
Thanks, Renzo